cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #2: compile fix, USB addition

mentaluproar, 2018-12-15:
Is it possible to fix this so it properly compiles on arm64?  It tries to compile on aarch64 on the rock64.  Forcing it to arm64 work, but would break the build on other platforms.  I don't have the skill to do a proper fix, just this hack.

Also, the TP-Link T4U sometimes includes an rtl8812BU chip, instead of an AU chip.  This is the only driver I've found that works.  Could you add the USB ID 2357,0115?

cilynx, 2018-12-16:
What do you mean by "forcing it to arm64"?  Are you doing something like `ARCH=amd64 make` or something else?  I don't have easy access to an arm64 build environment, so I'll need a bit more info to help here.

Re T4U, it's already included here:

https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/blob/c78ed5a4197fbaca3c94dcbe649f0dccf5203852/os_dep/linux/usb_intf.c#L233-L244

Are you running into issues with your device being detected?

mentaluproar, 2018-12-16:
I dint even thing of passing an argument.  I just swapped out the ARCH=$(whatevergoeshere) to arm64 in the makefile.

The device has a USB ID that isnt included in the usb....oh goddammit.

cilynx, 2018-12-16:
Passing `ARCH` as an environment variable should work, else it'll automagically be set to `uname -m | sed -e s/i.86/i386/`.  It looks like the Makefile is expecting `uname -m` to return "amd64" for that sort of system.  I assume `uname -m` returns "aarch64" on yours?  Assuming that's the case, try updating line 1033 of the Makefile to be `SUBARCH := $(shell uname -m | sed -e s/i.86/i386/ | sed -e s/aarch64/arm64/)` and let me know if it builds clean for you then with just a vanilla `make`.

https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/blob/c78ed5a4197fbaca3c94dcbe649f0dccf5203852/Makefile#L1033

As for your USB device, what does `lsusb` show?

mentaluproar, 2018-12-16:
passing arch did work, but it did not see my device.  When i got it working, I had that device ID under the 8812, not 8822.

cilynx, 2018-12-16:
I'm not aware of a V4.  Sucks that `lsusb` doesn't give you any more information than the ID.  Here are the T4U revisions to the best of my knowledge:
```
{USB_DEVICE(0x2357, 0x0101), RTL8812}, /* TP-Link - Archer T4U V1 (AC1200 Base Model) */
{USB_DEVICE(0x2357, 0x0103), RTL8812}, /* TP-Link - Archer T4UH V1 (AC1200 High Gain) */
{USB_DEVICE(0x2357, 0x010D), RTL8812}, /* TP-Link - Archer T4U V2 (AC1300 Base Model) */
{USB_DEVICE(0x2357, 0x010E), RTL8812}, /* TP-Link - Archer T4UH V2 (AC1300 High Gain) */
{USB_DEVICE(0x2357, 0x010F), RTL8812}, /* TP-Link - Archer T4UHP(UN) V1 (AC1300 High Power Wireless Dual Band USB Adapter) */
{USB_DEVICE(0x2357, 0x0115), RTL8822B}, /* TP-Link - Archer T4U V3 (AC1300 Wireless Dual Band USB Adapter) */
{USB_DEVICE(0x2357, 0x0122), RTL8812}, /* TP-Link - Archer T4UHP(US) V1 (AC1300 High Power Wireless Dual Band USB Adapter) */
{USB_DEVICE(0x2357, 0x012D), RTL8822B}, /* TP-Link - Archer T3U (AC1300 Mini Wireless MU-MIMO USB Adapter) */
```
Is [this](https://www.tp-link.com/my/download/Archer-T4U_V3.html) your adapter?  If not, you can flip through pictures of the different versions on that page.

mentaluproar, 2018-12-17:
That's it alright.

cilynx, 2018-12-17:
Alright -- let's go down this road a little further.  What do you get in your syslog when you plug the thing in?  You're looking for something vaguely similar to:
```
Dec 16 18:16:56 xps kernel: [3390056.534113] usb 1-1.4: new high-speed USB device number 47 using xhci_hcd
Dec 16 18:16:57 xps kernel: [3390056.646440] usb 1-1.4: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
Dec 16 18:16:57 xps kernel: [3390056.646444] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec 16 18:16:57 xps kernel: [3390056.646446] usb 1-1.4: Product: 802.11ac NIC
Dec 16 18:16:57 xps kernel: [3390056.646448] usb 1-1.4: Manufacturer: Realtek
Dec 16 18:16:57 xps kernel: [3390056.646449] usb 1-1.4: SerialNumber: 123456
```

cilynx, 2018-12-20:
A vendor not care about the product they push out the door?  Perish the thought!  I ordered one of the T4U v3 just for S&G.  It should be here on Friday and I can play around with which driver(s) work(s) and why.

mentaluproar, 2018-12-20:
Crack it open and check what chip it uses. Mine had me clueless until I opened it up and looked directly at the chip.

The arm64 make you threw in built right. It isn’t seeing any networks yet but I’m going to take a break from it for a few hours. It’s probably something I did.

cilynx, 2018-12-22:
Alrighty -- so the T4U v3 is [definitely an RTL8812BU chip](https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=3746160).  Adding 2357:0115 to either 8812AU or 8822BU driver makes the thing show up, but it never gets past "hardware disabled".  I've played around a bit with `rfkill` to no avail.  I've found a few folks who have other 8812BU powered adapters working with the 8822BU driver, but I'm not finding anyone who has this specific adapter working.  The CD that comes with the adapter has a text file saying to look online for the Linux driver.  Downloading the T4U v3 Linux driver straight from TP-LINK gets you rtl88x2BU v5.3.1_27282.20180419_COEX20180112-5959_beta which compiles and loads fine, but has exactly the same behavior (adapter shows up but perpetually "hardware disabled") as the others.  I'm going to have to play around with it a bit and see if I can convince it to work.

cilynx, 2018-12-22:
Scratch that -- the adapter is working 100% on Debian Unstable.  
```rcw@burner:~/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959$ cat /proc/version 
Linux version 4.19.0-1-amd64 (debian-kernel@lists.debian.org) (gcc version 8.2.0 (Debian 8.2.0-12)) #1 SMP Debian 4.19.9-1 (2018-12-16)
rcw@burner:~/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959$ gcc --version
gcc (Debian 8.2.0-13) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

rcw@burner:~/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959$ 
```

Now I just need to figure out why it doesn't work on Stretch.  You said you were having the same symptoms I see on Stretch, right?  The adapter is there, but no green LED and no networks?

mentaluproar, 2018-12-22:
No. The LED lit up and networks worked fine. The only issue is when using nmtui, it takes longer than normal to activate. I’m running on patched Linux 4.4 for the rock64. It’s on bionic so I don’t think it’s running Debian unstable. I might be able to try it on stretch too. I have some extra boards that I can flash to stretch easy tomorrow.

After family holiday stuff. 

That I found out about just this afternoon am expected for at 9 am. 

No I’m not bitchy about it.

cilynx, 2018-12-22:
...and scratch not working on Stretch.  The adapter doesn't seem to like the USB extension cable that I use on my main terminal.  Plugging it straight into one of the USB ports on the dock works fine.  My best guess is that it can't suck enough power through the extension cable as everything else I plug into that cable works fine.

Enjoy family holiday stuff =)

mentaluproar, 2019-01-02:
Holidays over, nuking and starting over on the rock64 project since I lost my place.  I cloned this again but the usb id for this weird adapter is still missing from usb_intf.c

cilynx, 2019-01-03:
Doesn't https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/blob/d953d367335e52a5f517fb2a04c4b45d42867eef/os_dep/linux/usb_intf.c#L240 have you covered?  If not, what ID are you looking for?

mentaluproar, 2019-01-03:
Didn’t work unless I put it in the first group. Let me nuke it and try again. The rock64 does really weird stuff sometimes if power is t exactly what it wants so I’m wondering if I had it plugged into an I sufficient supply last time I tested that issue.

cilynx, 2019-02-07:
Any luck?  My T4U v3 is working properly on multiple devices at this point.

mentaluproar, 2019-02-07:
I've been too busy lately to pick up the project that needs this. It's sort of fallen on the back burner.  I can confirm this driver does work, but for whatever reason, I had to do that edit mentioned above.  I don't have time to test if it still needs it right now though, sorry.

cilynx, 2019-02-07:
No worries -- I know how that goes.  I'll close the issue for now just for housekeeping, but feel free to reopen it in the future if it bubbles back up for you.

closed 2019-02-07 by cilynx

minusdavid, 2019-04-18:
> Alrighty -- so the T4U v3 is definitely an RTL8812BU chip. You can see it on the internal photos on the [FCC registration site](https://fcc.io/TE7/T4UV3). Adding 2357:0115 to either 8812AU or 8822BU driver makes the thing show up, but it never gets past "hardware disabled". I've played around a bit with `rfkill` to no avail. I've found a few folks who have other 8812BU powered adapters working with the 8822BU driver, but I'm not finding anyone who has this specific adapter working. The CD that comes with the adapter has a text file saying to look online for the Linux driver. Downloading the T4U v3 Linux driver straight from TP-LINK gets you rtl88x2BU v5.3.1_27282.20180419_COEX20180112-5959_beta which compiles and loads fine, but has exactly the same behavior (adapter shows up but perpetually "hardware disabled") as the others. I'm going to have to play around with it a bit and see if I can convince it to work.

I'm running Debian Stretch with a 4.9.0.8 Kernel and using the 5.3.1 beta driver from https://www.tp-link.com/au/support/download/archer-t4u/#Driver. I can compile it and on the surface it looks like it's working... except the password I put in for the network never works...

minusdavid, 2019-04-18:
Tried the steps mentioned at https://askubuntu.com/a/1067500 but still same problem.

Looking at syslog and it says it's the WRONG_KEY which can't be right. So it makes me think that the drivers aren't handling the password correctly.

minusdavid, 2019-04-18:
Apr 18 22:18:03 icarus kernel: [ 2466.516105] RTW: WARN _beamforming_leave: 8c:f5:a3:aa:9f:78 is neither beamforming ee or er!!
Apr 18 22:18:03 icarus kernel: [ 2466.516110] RTW: cfg80211_rtw_del_key(wlx503eaaae571c) key_index=0, addr=          (null)
Apr 18 22:18:03 icarus kernel: [ 2466.516113] RTW: cfg80211_rtw_del_key(wlx503eaaae571c) key_index=1, addr=          (null)
Apr 18 22:18:03 icarus kernel: [ 2466.516114] RTW: cfg80211_rtw_del_key(wlx503eaaae571c) key_index=2, addr=          (null)
Apr 18 22:18:03 icarus kernel: [ 2466.516115] RTW: cfg80211_rtw_del_key(wlx503eaaae571c) key_index=3, addr=          (null)
Apr 18 22:18:03 icarus kernel: [ 2466.516117] RTW: cfg80211_rtw_del_key(wlx503eaaae571c) key_index=4, addr=          (null)
Apr 18 22:18:03 icarus kernel: [ 2466.516118] RTW: cfg80211_rtw_del_key(wlx503eaaae571c) key_index=5, addr=          (null)
Apr 18 22:18:03 icarus kernel: [ 2466.516424] RTW: rtw_ndev_notifier_call(wlx503eaaae571c) state:4
Apr 18 22:18:03 icarus wpa_supplicant[699]: wlx503eaaae571c: CTRL-EVENT-DISCONNECTED bssid=8c:f5:a3:aa:9f:78 reason=3
Apr 18 22:18:03 icarus wpa_supplicant[699]: wlx503eaaae571c: WPA: 4-Way Handshake failed - pre-shared key may be incorrect
Apr 18 22:18:03 icarus wpa_supplicant[699]: wlx503eaaae571c: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="AndroidAP" auth_failures=1 duration=10 reason=WRONG_KEY

minusdavid, 2019-04-18:
Anyone else having issues like this?

Apr 18 22:22:55 icarus NetworkManager[625]: <info>  [1555590175.0381] device (wlx503eaaae571c): Activation: (wifi) disconnected during association, asking for new key
Apr 18 22:22:55 icarus NetworkManager[625]: <info>  [1555590175.0400] device (wlx503eaaae571c): state change: config -> need-auth (reason 'supplicant-disconnect') [50 60 8]
Apr 18 22:22:55 icarus NetworkManager[625]: <info>  [1555590175.1317] device (wlx503eaaae571c): supplicant interface state: disconnected -> inactive
Apr 18 22:22:56 icarus NetworkManager[625]: <warn>  [1555590176.5032] device (wlx503eaaae571c): User canceled the secrets request.
Apr 18 22:22:56 icarus NetworkManager[625]: <info>  [1555590176.5032] device (wlx503eaaae571c): state change: need-auth -> failed (reason 'no-secrets') [60 120 7]
Apr 18 22:22:56 icarus NetworkManager[625]: <warn>  [1555590176.5039] device (wlx503eaaae571c): Activation: failed for connection 'AndroidAP'
Apr 18 22:22:56 icarus NetworkManager[625]: <info>  [1555590176.5048] device (wlx503eaaae571c): state change: failed -> disconnected (reason 'none') [120 30 0]

minusdavid, 2019-04-21:
> I'm running Debian Stretch with a 4.9.0.8 Kernel and using the 5.3.1 beta driver from https://www.tp-link.com/au/support/download/archer-t4u/#Driver. I can compile it and on the surface it looks like it's working... except the password I put in for the network never works...

It turns out the solution is here: https://wiki.debian.org/WiFi/HowToUse#WiFi_can_scan.2C_but_not_connect_using_NetworkManager_.28Debian_9_Stretch.29

I added the following to /etc/NetworkManager/NetworkManager.conf and did a service NetworkManager restart and all was well afterwards:

[device]
wifi.scan-rand-mac-address=no

Polve, 2019-04-21:
I've the same problem on Ubuntu 19.04 Disco, I checked the hint by @minusdavid but unfortunately it doesn't help since the setting is already that way :-(

zFlorent, 2019-08-10:
Hello,

I compiled rtl8822bu drivers from [jeremyb31](https://github.com/jeremyb31/rtl8822bu) on opensuse 15.1 (Linux version 4.12.14-lp151.28.10-default (geeko@buildhost) (gcc version 7.4.0 (SUSE Linux) ) on a x86-64 PC platform.
I had to modify Makefile in order to successfully compile by adding : 
EXTRA_CFLAGS += -Wno-error=incompatible-pointer-types

My device is a TP-link AC1300 Archer T3U (lsusb -> ID 2357:012d)
{USB_DEVICE(0x2357, 0x012D), RTL8822B}, /* TP-Link - Archer T3U (AC1300 Mini Wireless MU-MIMO USB Adapter) */

The driver manage to connect to my 2GHz wifi network, but with poor speed (Max 50kB/s). And is not able to use 5GHz network (Configuring the interface -> disconnected).

Any idea why ?
I took this adapter because it was supposed to be linux compatible... :-(

BR,
Florent