cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #4: Is it possible to set txpower?

MichaelHsu170, 2018-12-23:
I'm using a rtl8822B wifi dongle.
"iw list" confirms that txpower at 2.4G band could go up to 20dbm. Although I tried to set txpower with iw command, it seems to be fixed at 12dbm.
Is it possible to raise the txpower with this driver?

cilynx, 2019-02-07:
This is not something I've played with.  There are [a couple TX_POWER configuration options in the Makefile](https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/blob/master/Makefile#L889-L895).  Give them a shot and let us know how it goes.

cilynx, 2019-03-13:
Closing due to lack of motion.  Feel free to reopen if you test the config options and have further comments / questions.

closed 2019-03-13 by cilynx