cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #5: kernel 4.20

joksik, 2018-12-26:
hello after i update my kernel to kernel 4.20 i cant build 88x2bu drivers. in my logs i have this:

DKMS make.log for rtl88x2bu-5.3.1 for kernel 4.20.0-pf1 (x86_64)
śro, 26 gru 2018, 17:02:39 CET
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.20.0-pf1/build M=/var/lib/dkms/rtl88x2bu/5.3.1/build  modules
make[1]: Wejście do katalogu '/usr/lib/modules/4.20.0-pf1/build'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/wifi_regd.o
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_get_systime_us’:
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_cfg80211.c:691:2: error: implicit declaration of function ‘get_monotonic_boottime’; did you mean ‘getboottime’? [-Werror=implicit-function-declaration]
  get_monotonic_boottime(&ts);
  ^~~~~~~~~~~~~~~~~~~~~~
  getboottime
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/ioctl_mp.o
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:291: /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/ioctl_cfg80211.o] Błąd 1
make[2]: *** Oczekiwanie na niezakończone zadania....
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_write_reg’:
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c:46:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  char input[wrqu->length + 1];
  ^~~~
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_read_reg’:
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c:132:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  char input[wrqu->length + 1];
[... 63 lines cut ...]
  char input[wrqu->length];
  ^~~~
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_SetRFPath’:
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c:1273:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  char input[wrqu->length];
  ^~~~
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_switch_rf_path’:
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c:1322:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  char input[wrqu->length];
  ^~~~
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_QueryDrv’:
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c:1371:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  char input[wrqu->data.length];
  ^~~~
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_PwrCtlDM’:
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c:1400:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  u8  input[wrqu->length];
  ^~
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_hwtx’:
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c:2050:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  u8  input[wrqu->data.length];
  ^~
make[1]: *** [Makefile:1563: _module_/var/lib/dkms/rtl88x2bu/5.3.1/build] Błąd 2
make[1]: Opuszczenie katalogu '/usr/lib/modules/4.20.0-pf1/build'
make: *** [Makefile:1986: modules] Błąd 2

cilynx, 2018-12-26:
Looks like `get_monotonic_boottime` has been deprecated in 4.20.  Will need to do some testing, but we can probably replace it with `ktime_get_boottime`.  Stay tuned.

cilynx, 2018-12-26:
I just pushed a fix to a [new branch](https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/tree/fix/kernel4.20), but I don't have a 4.20 system readily available to test.  Can you build from that branch and let me know how it goes?

PR #6 (2018-12-27): Fix/kernel4.20
Confirmed by @joksik -- Fixes #5
closed 2018-12-27 by cilynx