cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #8: Is AP supported?

alvaroga91, 2019-02-05:
Thanks for this! 

I come from here as I'm interested in using a Wi-Fi module with RTL8822: https://github.com/brandon-bailey/rtl8822bu

The limitation was that AP was not supported. Maybe (hopefully), is a stupid question, but is it supported in this one?

cilynx, 2019-02-07:
I've honestly not tried it.  `iw` says that `AP` is a supported mode:
```
Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
                 * P2P-device
```

Give it a try and let us know how it goes =)

cilynx, 2019-03-13:
Closing due to lack of motion.  Feel free to reopen if you've tested and have further comments / questions.

closed 2019-03-13 by cilynx

cilynx, 2019-03-15:
Inspired by @Philippe12's [fork](https://github.com/Philippe12/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959), I made a [couple tweaks]( https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/commit/57fc88352c60487d9d0c942b7dadbb4c2c8b5a79) to get this working and documented in the [README.md](https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959#raspberry-pi-access-point).  Confirmed operational on a 3B+ w/ a fresh install of Raspbian.

PR #14 (2019-03-15): Fix/raspi ap mode
Fixes #8