cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #9: Monitor mode virtual interface creation

nt-o, 2019-02-25:
Referencing https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.2.4.1_22719_COEX20170518-4444.20170613/issues/2

Specifically:
>Playing around with kismet, it fails to create a mon VIF, but successfully falls back to putting the real interface into Monitor Mode without hanging or jacking up the network stack. Unfortunately, it doesn't ever receive any packets / see any networks or clients.
>
>I'm out of time to dig into this today. If you feel like poking around with things further, I'd be interested to hear what you find.

My system:
Linux kali 4.19.0-kali1-amd64 #1 SMP Debian 4.19.13-1kali1 (2019-01-03) x86_64 GNU/Linux

Compiling the driver with the monitor flag in the makefile set to "y" allows me to use iw to set the interface to monitor mode and capture packets using kismet.  The problem with airmon-ng persists.  I don't know anything about anything but it appears that there might be a relevant TODO on line 4497 of /os_dep/linux/ioctl_cfg80211.c:

```c
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 6, 0))
static struct wireless_dev *
#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 38)) || defined(COMPAT_KERNEL_RELEASE)
static struct net_device *
#else
static int
#endif
	cfg80211_rtw_add_virtual_intf(
		struct wiphy *wiphy,
		#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 7, 0))
		const char *name,
		#else
		char *name,
		#endif
		#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 1, 0))
		unsigned char name_assign_type,
		#endif
		enum nl80211_iftype type,
		#if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 12, 0))
		u32 *flags,
		#endif
		struct vif_params *params)
{
	int ret = 0;
	struct wireless_dev *wdev = NULL;
	struct net_device *ndev = NULL;
	_adapter *padapter;
	struct dvobj_priv *dvobj = wiphy_to_dvobj(wiphy);

	rtw_set_rtnl_lock_holder(dvobj, current);

	RTW_INFO(FUNC_WIPHY_FMT" name:%s, type:%d\n", FUNC_WIPHY_ARG(wiphy), name, type);

	switch (type) {
	case NL80211_IFTYPE_MONITOR:
		padapter = wiphy_to_adapter(wiphy); /* TODO: get ap iface ? */
		ret = rtw_cfg80211_add_monitor_if(padapter, (char *)name, &ndev);
		if (ret == 0)
			wdev = ndev->ieee80211_ptr;
		break;
```

katzz0, 2019-03-27:
It seems the same TODO comment here https://github.com/aircrack-ng/rtl8812au/blob/v5.2.20/os_dep/linux/ioctl_cfg80211.c#L4264 but the driver supports monitor mode and frame injection.

Naive solution: compare both repos and add monitor mode fixes here.

yufalcon, 2020-10-11:
> It seems the same TODO comment here https://github.com/aircrack-ng/rtl8812au/blob/v5.2.20/os_dep/linux/ioctl_cfg80211.c#L4264 but the driver supports monitor mode and frame injection.
> 
> Naive solution: compare both repos and add monitor mode fixes here.

can use fluxion?