cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #10: I need a driver Module for my Synology nas

SoulInfernoDE, 2019-03-02:
Hi,
as cheap usb 3 wifi dongles aren't supported any more for many synology devices i need to compile this driver for DSM 6.2 with braswell architecture: https://sourceforge.net/projects/dsgpl/files/DSM%206.1%20Tool%20Chains/Intel%20x86%20Linux%203.10.102%20%28Braswell%29/
Is this doable or do i need to modify something i the make file?

cilynx, 2019-03-03:
[This](http://www.wolfteck.com/2019/03/02/packaging_rtl88x2bu_for_synology_nas/) should get you most of the way there.  You may have to play around with `CONFIG_IOCTL_CFG80211` and/or `mac80211.ko`.  See the build section of the write-up and google the warnings and you'll find useful discussion.

Let me know if / how you get it working.  If it does actually work, I'll commit the platform to the repo to cut down on the steps required.

SoulInfernoDE, 2019-03-04:
Hey,
cilynx. WOW! This is somewhat helpful! Do you have a donation link or paypal mail. I want to donate to you because you helped/answered so fast.. ...didn't had time yet to go through but will do it next week hopefully after my shift period has ended. Definitly let you know any results and errors i face then. I am going to install the spk and give it a try but from what i have read so far there are some more tweaks in the udev scripts needed. Here is a tutorial i had found before asking that got me near to goal i guess: [https://und3ath.github.io/2018/03/06/Synology-usbDongle-integration.html](https://und3ath.github.io/2018/03/06/Synology-usbDongle-integration.html)
Your manual/tutorial you wrote will be superior on the internet. Nice work so far!!!

cilynx, 2019-03-04:
Ha -- I falsely assumed that since the kernel is `x86_64` that `x64` was the right platform.  Shows I really don't have any idea what I'm doing here.  =)  Perhaps I should have checked `EnvDeploy -v 6.2 --list` before making that assumption.  Anyways -- I've [updated](https://github.com/cilynx/cilynx.github.io/commit/241c1cf1851b14b5d2c53ca446dc0548d44edba4) the [writeup](http://www.wolfteck.com/2019/03/02/packaging_rtl88x2bu_for_synology_nas/) and here are the `braswell` SPKs:

[rtl88x2bu-braswell-5.3.1_27678.20180430_COEX20180427-5959.tar.gz](https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/files/2927226/rtl88x2bu-braswell-5.3.1_27678.20180430_COEX20180427-5959.tar.gz)

If these don't work, you're going to have to fiddle around with `cfg80211` / `mac80211` / etc and figure out what the missing dependencies are.

SoulInfernoDE, 2019-03-04:
_I'm so speechless how fast you are. It did install but i can't find the rtl88x2bu.ko file yet in /lib/modules/. Will look through next week after my shift period. This is so awesome! ...so genius. Don't forget to send me your donation link._
[image]

### Preview of installed app according to the synology app store:
[image]

As mentioned it installed right away without error messages from the gui but couldn't find the module yet. Here is a list of my "**/lib/modules**"  filtered with _rtl - cfg - mac_

**`root@ds716plus: /lib/modules#`**`ls rtl* cfg* mac*`
[`cfg80211.ko`](https://owncloud.damnserver.com/index.php/s/V2nWUZYT72ktA0S)
[`mac80211.ko`](https://owncloud.damnserver.com/index.php/s/V2nWUZYT72ktA0S)
`macvlan.ko`
`rtl8187.ko`
`rtl8192c-common.ko`
`rtl8192cu.ko`
`rtl_usb.ko`
`rtlwifi.ko`

_I've added the two files that were mentioned while you were compiling the driver._

cilynx, 2019-03-04:
I'm making this up as I go, but as far as I can tell, SPKs are unpacked to `SYNOPKG_PKGDEST` and if you want things like kernel modules to go to their normal places, you have to do so via those scripts like `postinst`.

[USBSerialDrivers_4.0.2.spk](http://www.synorepo.tk/packages/USBSerialDrivers_4.0.2.spk) has a good example of a more complex startup script that does some module copying.  If you pull down the SPK and untar it, you'll find the `scripts` directory at the root of the archive.  There's some substantial quality discussion [on the Synology forum](https://forum.synology.com/enu/viewtopic.php?t=82843) as well.

For the relatively simple use case, the `start-stop-status` script I baked into the SPK should `insmod` the module when you click Run in the Package Manager and `rmmod` the module when you click Stop.  This is assuming that [Synology's example `start-stop-status`](https://github.com/SynologyOpenSource/HelloKernel/blob/master/scripts/start-stop-status) works as expected.

There's a great writeup on SPKs in general [here](https://forum.synology.com/enu/viewtopic.php?t=10807).

I've gone about as far as I can go without having the hardware to poke at and see if things work, but I'm very interested to see if you get this working and I'm happy to give suggestions if you get stuck.

Re donation, please don't feel obliged as I do this stuff for fun, but if you're so inclined:

[image]

SoulInfernoDE, 2019-03-04:
Donation is out to you,
it should be worshipped a bit at least :+1: 

_->> Yes i was aware of those packages (USBSerialDrivers; USBTools) originally designed by: jadahl.com from domoticz software. And was playing around earlier with it a bit. Unfortunately they do not add support for 8812bu only for 8812au and 8821au. (The original domoticz package -not the serial driver package - obtained from jadahl.com)._

I really appreciate what you have done so far !! - Will follow up next week as fast as i can... :+1:  :+1: 

-> Nice to see that we have mostly the same level of information already :-)

SoulInfernoDE, 2019-03-13:
I have successfully compiled the driver module. I will add details later as i am in hurry.

copy module to "/lib/modules" on your NAS and then "chmod 644 88x2bu.ko" it. After that "insmod /lib/modules/88x2bu.ko".
Check for wifi with: iwconfig
wlan0 should show you a realtek wifi.
As for the gui i am working on a solution. Couldn't get it to be visible in the DSM gui yet.

**DSM 6.2 - Braswell Realtek 88X2BU driver module**
[88x2bu.ko.tar.gz](https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/files/2962511/88x2bu.ko.tar.gz)

cilynx, 2019-03-13:
Great to hear you got it working.  Looking forward to details -- I'll happily add the NAS as a first-class target once we've got the details worked out.

SoulInfernoDE, 2019-03-22:
The driver attached above was compiled with manually downloading the tool chain and the kernel source.

I have set-up a development environment on my second ssd. I've used "lmde" (Linux Mint Debian Edition) - because i like the start menu at the bottom left.
I then followed the tutorial first that i had found for the older arm diskstation and did set all the recommended variables
'CFLAGS LDFLAGS RANLIB LD CC LD_LIBRARY_PATH ARCH CROSS_COMPILE KSRC'
Then i followed your writeup while reading official docs from synology.

It did compile successfully and didn't throw any undefined! messages but i think that was because i used only
'ifeq ($(CONFIG_PLATFORM_SYNOLOGY_BRASWELL), y)
EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
ARCH := x86_64
endif' in the Makefile

Today i tried to add cfg80211 and the sta event parameter to the makefile config - then it throwed the warnings again. I'm not sure yet what cfg80211 does or for what it's needed for. I did search the internet about it but couldn't understood it yet or not fully understood it yet. If that is important to use and you have more experience with cfg80211 would you be so kind to tell me what you know about this files/parameters? - If i am correct i think it has something to do with programming/compiling the switches and simililar things into the driver where the user could pass parameters correctly to the driver so that the driver understood the calls? ...if that is the case or if the parameter and the files are really needed then this could be the culprint why my web gui of the diskstation manager software couldn't recognize the stick. I used ifconfig wlan0 up and the sticks led started to blink. But it feels like im missing something - because i am not able to get the gui recognize it. If i click on system settings -> system information center i see under external devices: 802.11ac NIC Realtek Semiconductor so the nas recognize it in some way but not fully yet. If i click on "WLAN" in the system settings it says something like "please put a wifi stick in your disk station". This is the tab where you normally configure all the things.
As i have seen in a config file of the diskstation some rtl-driver modules have {COMPAT} or {CFG80211} added behind {RTL8812AU}. Im not sure if this is a describer for what needs to be loaded or if that is a config that can be preloaded from already existing modules. I will post the files content later on. Maybe we can figure all the details out. I'm excited and i like to learn as much as i could. Only my shift job cuts my learning phases off. Sometimes ten days per shift in a row. That's why i need some time to get back to work :-)  ..sorry for that.

There is also a 3rd party community called synocommunity which made something like a compile toolkit also. This is something i want to try later on also.

ablkbs, 2021-01-12:
Dear All,
I'm a newbie for linux after my 35th year old and I'm trying to learn linux and coding. This is just a new hobby for me but the problem is I'm sticking on the things I'm trying.

I have a Synology DSM216j for home use and trying to run my Asus AC-53 Nano which has a RTL8812BU chipset on NAS. I have found several tutorials but I couldn't find a solution to install the driver into NAS. On some versions I have faced problems with the dirver versions, on some versions about the files.

I will be very happy if you could help me to install the driver. It is too hard for me to do it with my current knowledge.

~$ uname -a
Linux Synology_AB 3.10.105 #25426 SMP Mon Dec 14 18:44:29 CST 2020 armv7l GNU/Linux synology_armada38x_ds216j

Asus has its own linux drivers within the package on the link: https://dlcdnets.asus.com/pub/ASUS/wireless/USB-AC53_Nano/DR_USB_AC53_Nano_v1016_US.zip

GPL Source and Tool Chain: https://sourceforge.net/projects/dsgpl/

Thanks in advance for your help.

camelcx, 2021-03-29:
> The driver attached above was compiled with manually downloading the tool chain and the kernel source.
>
> I have set-up a development environment on my second ssd. I've used "lmde" (Linux Mint Debian [...]

Hi, 

I'm in a very similar situation compiling the driver for the same chipset. Only difference is the Synology platform (evansport in my case). I have come to a similar stage where the kernel module does compile and load. However, it does not actually 'work' or gets recognised by the gui. Have you found a solution regarding this? Or maybe point me to the correct direction? (I too have to remove the CFG80211 option for the module to load, and have no idea if that is a required option...)

Many thanks.