cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #11: Linux Kernel 5.0 support possible?

SoulInfernoDE, 2019-03-09:
DKMS make.log for rtl88x2bu-5.3.1 for kernel 5.0.0-050000-generic (x86_64)
Sa 9. Mär 10:59:45 CET 2019
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.0.0-050000-generic/build M=/var/lib/dkms/rtl88x2bu/5.3.1/build  modules
make[1]: Verzeichnis „/usr/src/linux-headers-5.0.0-050000-generic“ wird betreten
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_proc.o
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/rtw_android.c: In function ‘rtw_android_priv_cmd’:
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/rtw_android.c:660:62: error: macro "access_ok" passed 3 arguments, but takes just 2
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
                                                              ^
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/rtw_android.c:660:7: error: ‘access_ok’ undeclared (first use in this function)
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
       ^~~~~~~~~
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/rtw_android.c:660:7: note: each undeclared identifier is reported only once for each function it appears in
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_rhashtable.o
scripts/Makefile.build:276: recipe for target '/var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_android.o' failed
make[2]: *** [/var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_android.o] Error 1
make[2]: *** Auf noch nicht beendete Prozesse wird gewartet …
Makefile:1553: recipe for target '_module_/var/lib/dkms/rtl88x2bu/5.3.1/build' failed
make[1]: *** [_module_/var/lib/dkms/rtl88x2bu/5.3.1/build] Error 2
make[1]: Verzeichnis „/usr/src/linux-headers-5.0.0-050000-generic“ wird verlassen
Makefile:1995: recipe for target 'modules' failed
make: *** [modules] Error 2

cilynx, 2019-03-09:
Checkout the new [linux-5.0 branch](https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/tree/linux-5.0) and lmk if that works for you.  I tested build and module load on an Ubuntu 18.10 VM with Linux version 5.0.0-050000-generic, but VirtualBox's USB passthrough isn't liking me right now, so I can't confirm functionality.

SoulInfernoDE, 2019-03-13:
<pre><span style="background-color:#FFFFFF"><font color="#3F3F3F">  GNU nano 2.9.3    /var/lib/dkms/rtl88x2bu/5.3.1/build/make.log                </font></span>

DKMS make.log for rtl88x2bu-5.3.1 for kernel 5.0.0-050000-generic (x86_64)
Mi 13. Mär 12:03:00 CET 2019
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.0.0-050000-generic/build M=/var/lib/dkms/rtl88x2bu/5.3.1/build  modules
make[1]: Verzeichnis „/usr/src/linux-headers-5.0.0-050000-generic“ wird betreten
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/ioctl_mp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/hal/hal_intf.o
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/rtw_android.c: In function ‘rtw_android_priv_cmd’:
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/rtw_android.c:660:62: error: macro "access_ok" passed 3 arguments, but takes just 2
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
                                                              ^
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/rtw_android.c:660:7: error: ‘access_ok’ undeclared (first use in this function)
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
       ^~~~~~~~~
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/rtw_android.c:660:7: note: each undeclared identifier is reported only once for each function it appears in
scripts/Makefile.build:276: recipe for target '/var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_android.o' failed
make[2]: *** [/var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/rtw_android.o] Error 1
make[2]: *** Auf noch nicht beendete Prozesse wird gewartet …
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_write_reg’:
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c:46:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  char input[wrqu->length + 1];
[... 70 lines cut ...]
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c:1273:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  char input[wrqu->length];
  ^~~~
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_switch_rf_path’:
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c:1322:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  char input[wrqu->length];
  ^~~~
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_QueryDrv’:
[... cut ...]
/usr/src/rtl88x2bu-5.3.1/os_dep/linux/ioctl_mp.c:2050:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  u8  input[wrqu->data.length];
  ^~
Makefile:1553: recipe for target '_module_/var/lib/dkms/rtl88x2bu/5.3.1/build' failed
make[1]: *** [_module_/var/lib/dkms/rtl88x2bu/5.3.1/build] Error 2
make[1]: Verzeichnis „/usr/src/linux-headers-5.0.0-050000-generic“ wird verlassen
Makefile:1995: recipe for target 'modules' failed
make: *** [modules] Error 2
<span style="background-color:#FFFFFF"><font color="#3F3F3F">^G</font></span>

cilynx, 2019-03-13:
Thanks for confirming!  I'll pull the update into master.  The warnings about VLAs are safe to ignore as they are valid extensions in `gnu89`.  I'm not sure why Ubuntu includes `-Wvla` to warn about VLAs by default while other distros don't, but that appears to be the case.

closed 2019-03-13 by cilynx

cilynx, 2019-03-15:
Correcting myself here -- `-Wvla` was [added to the kernel in 4.20](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=0bb95f80a38f82884693194ea720e9cca5e12ada) -- has nothing to do with Ubuntu vs other distros.

PR #12 (2019-03-13): Upstream `access_ok` no longer takes `type`
Fixes #11