cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #17: TP-Link Archer T4U AC1300 V3

vav4eg, 2019-05-02:
Hi, I tested the driver on TP-Link Archer T4U AC1300 V3. I was very pleased. The driver was going without errors. Remarkably goes into monitoring mode with the command - sudo ip link set wlan0 .... Scans networks (2.4GHz and 5 GHz), intercepts handshakes, everything happens under the wlan0 interface (as if the wlan0mon interface was created)
Identified deficiencies:
- will not receive reply packets in reaver ([+] Sending authentication request, [!] WARNING: Receive timeout occurred)
- does not enter the monitoring mode with the command - air ...... start wlan0

I would be very blogging for any advice on how to resolve the identified shortcoming

Linux version 5.0.11

mkylasoto, 2019-06-13:
Hello, can you explain how you were able to use this driver? I keep getting error 2 after attempting to make. 

root@kali:~/Downloads/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959-master# make
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.19.0-kali4-amd64/build M=/root/Downloads/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959-master  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-kali4-amd64'
/bin/sh: 1: bc: not found
  CC [M]  /root/Downloads/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959-master/core/rtw_debug.o
/root/Downloads/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959-master/core/rtw_debug.c: In function ‘dump_drv_version’:
/root/Downloads/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959-master/core/rtw_debug.c:45:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
                                                              ^
/root/Downloads/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959-master/core/rtw_debug.c:45:62: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
/root/Downloads/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959-master/core/rtw_debug.c:45:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
/root/Downloads/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959-master/core/rtw_debug.c:45:62: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-4.19.0-kali4-common/scripts/Makefile.build:309: /root/Downloads/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959-master/core/rtw_debug.o] Error 1
make[3]: *** [/usr/src/linux-headers-4.19.0-kali4-common/Makefile:1535: _module_/root/Downloads/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959-master] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-kali4-amd64'
make: *** [Makefile:1999: modules] Error 2

vav4eg, 2019-06-13:
Hello, I collected the driver in Ubuntu, but based on the logs, then you use Kali. I do not know how to solve your problem, but in the forums of my country, it is advised to use the - configure --disable-werror switch.
Try writing a driver developer, maybe they will help you.

vav4eg, 2019-06-13:
They also write that there is a complaint about the absence of **bc**, so you need to install it with the **install** command.

mkylasoto, 2019-06-13:
Ah, okay. Thank you! I’ll try the driver developer.