cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #22: Fails to connect: Ubuntu 18.04 with ANEWISH adapter on front USB ports only!?

A2Zog, 2019-05-25:
Was running [ANEWISH AC1200 USB](https://www.amazon.com/gp/product/B07FGX8LBF/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1) fine on Ubuntu 18.04 (4.18.0-20-generic) with MSI motherboard/AM3/Athlon and USB 3 add-on card and front panel ports.

Upgraded to ASRock motherboard/AM4/Ryzen) with same front panel ports.

If I use the motherboards rear USB3 IO ports, all is well. If I use the front panel ports, nothing connects. Seems to be same behavior for 2.4 and 5GHz. A real head scratcher!

dmesg shows (to my eye) the same info until the successful port reports **RTW: OnBeacon: beacon keys ready** while the unsuccessful port reports **RTW: link_timer_hdl:no beacon while connecting and never recovers**. (See attached.)

Note, on same hardware, Windows 10 runs in both configurations, so I don't think it is the motherboard header or the front panel, but I could be wrong. USB3 HDD works fine in same front panel ports.

I'll also note that lsusb shows 480M connection speed in both cases.

Any thoughts? How may I assist debugging?

[dmesg_88x2bu_front_bad.txt](https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/files/3219173/dmesg_88x2bu_front_bad.txt)
[dmesg_88x2bu_rear_good.txt](https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/files/3219174/dmesg_88x2bu_rear_good.txt)

Additional note: same behavior with [FomalhautWeisszwerg/rtl8822bu](https://github.com/FomalhautWeisszwerg/rtl8822bu/issues/27) driver

RandyLoeb, 2019-07-09:
Thanks...though I don't have a solution, I was unable to connect and after I read your post it occurred to me to try another port...and now it connects!

A2Zog, 2019-07-10:
Still not working, but I have seen other weirdnesses with my MB USB 3 header, so I'm closing this.

closed 2019-07-10 by A2Zog