cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #23: Error!

Montana552, 2019-05-30:
Hi,

I'm receiving the following. Any ideas?
```

root@kali:~/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959# sudo dkms build -m rtl88x2bu -v ${VER}

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j2 KERNELRELEASE=4.19.0-kali4-amd64 KVER=4.19.0-kali4-amd64 src=/usr/src/rtl88x2bu-5.3.1.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.0-kali4-amd64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.3.1/build/make.log for more information.

```

cilynx, 2019-05-31:
Can you provide the make.log referenced in the error?

adrium, 2019-06-02:
Try putting `make` in single quotes in the dkms.conf and try again. Like so:

    MAKE[0]="'make' KVER=$kernelver src=$source_tree/rtl88x2bu-$PACKAGE_VERSION"
    CLEAN="'make' clean"

prosinholi, 2019-07-01:
Same error with a newer kernel version:

`Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.0-kali5-amd64 KVER=4.19.0-kali5-amd64 src=/usr/src/rtl88x2bu-5.3.1....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.0-kali5-amd64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.3.1/build/make.log for more information.
`

Tried to add the make inside simple quotes and got the error below:

`KMS make.log for rtl88x2bu-5.3.1 for kernel 4.19.0-kali5-amd64 (x86_64)
Mon 01 Jul 2019 04:53:27 PM EDT
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.19.0-kali5-amd64/build M=/var/lib/dkms/rtl88x2bu/5.3.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-kali5-amd64'
/bin/sh: 1: bc: not found
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o
/usr/src/rtl88x2bu-5.3.1/core/rtw_debug.c: In function ‘dump_drv_version’:
/usr/src/rtl88x2bu-5.3.1/core/rtw_debug.c:45:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
                                                              ^
/usr/src/rtl88x2bu-5.3.1/core/rtw_debug.c:45:62: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
/usr/src/rtl88x2bu-5.3.1/core/rtw_debug.c:45:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
/usr/src/rtl88x2bu-5.3.1/core/rtw_debug.c:45:62: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme.o
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-4.19.0-kali5-common/scripts/Makefile.build:308: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-headers-4.19.0-kali5-common/Makefile:1539: _module_/var/lib/dkms/rtl88x2bu/5.3.1/build] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-kali5-amd64'
make: *** [Makefile:1999: modules] Error 2`

prosinholi, 2019-07-02:
Error was that bc was missing. After installing it the build completed.

apt-get install bc

closed 2019-07-10 by cilynx