cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #24: Slow speed on 5GHz with T4Uv3

adrium, 2019-06-02:
Hi, I tried your driver for the [TP-Link Archer T4Uv3](https://www.tp-link.com/us/home-networking/usb-adapter/archer-t4u/).

It compiles and device works of sorts. I can connect to my 5GHz network, but I get miserable speed. Sometimes less then one Mbps download. Interestingly, the upload was around 80 Mbps in one test.

I also tried to use the [official driver by TP-Link](https://www.tp-link.com/us/support/download/archer-t4u/) and applied your fixes. [Here is a repo of the merge](https://github.com/adrium/rtl88x2bu).

I played with the `rtw_power_mgnt` and `rtw_smart_ps` parameters without any luck.

The results are the same: low speeds.

Do you have any advice on improving the situation?

Niproblema, 2019-07-17:
`/etc/NetworkManager/conf.d/default-wifi-powersave-on.conf` from 3 (enabled) to 2 (disabled).
Helps me nearly match windows drivers speed

adrium, 2019-07-21:
Thanks, that helped!

I also tried the parameters `rtw_power_mgnt=0 rtw_ips_mode=0` which gave very good results as well.

closed 2019-07-21 by adrium