cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #27: USB 3.0 support?

lemonahmas, 2019-07-02:
Is there any way to make the rtl8812bu usb3.0 stick really use usb 3.0?

rrafal1337, 2019-07-09:
[ 2462.775697] usb 8-2: new high-speed USB device number 4 using xhci_hcd
[ 2462.967071] usb 8-2: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[ 2462.967073] usb 8-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2462.967074] usb 8-2: Product: 802.11ac NIC
[ 2462.967075] usb 8-2: Manufacturer: Realtek
[ 2462.967076] usb 8-2: SerialNumber: 123456

 lsusb -t
/:  Bus 11.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 10.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 09.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    |__ Port 2: Dev 4, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M

I believe it is not related to driver. This device utilizes usb 2.0 speed. My Spacetronik AC-1200 looks like usb 3 because of blue connector but it is really usb 2.0.

dmarkey, 2019-09-04:
`sudo modprobe 88x2bu rtw_switch_usb_mode=1`

dmarkey, 2019-09-04:
```
➜  ~ lsusb -t                 
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 5000M
    |__ Port 2: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
```

zuzia-dev, 2019-11-28:
Port USB 3.0.
```
modprobe 88x2bu rtw_switch_usb_mode=1
lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 5: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 5000M
```

```
lsusb -v
Bus 004 Device 002: ID 0bda:b812 Realtek Semiconductor Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xb812 
  bcdDevice            3.00
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC
 
    **MaxPower              126mA**

      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
```
And disconnects every few seconds.

Port USB 2.0.
```
modprobe 88x2bu
lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
    |__ Port 5: Dev 7, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
    |__ Port 6: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M

lsusb -v
Bus 003 Device 007: ID 0bda:b812 Realtek Semiconductor Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xb812 
  bcdDevice            2.10
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC

    **MaxPower              500mA**

  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x0006
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
```
It works fine.

wonko6x9, 2019-11-30:
I would like to bump this issue.  I am working towards around 100 of cards using this driver, and they seem to be dropping and picking back up in much the same way listed.  It appears they are also only running at USB2 speeds.

/:  Bus 10.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 09.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
        |__ Port 2: Dev 4, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
        |__ Port 3: Dev 5, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
        |__ Port 4: Dev 6, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
        |__ Port 2: Dev 4, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
        |__ Port 4: Dev 5, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 4: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 480M
    |__ Port 5: Dev 3, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
    |__ Port 6: Dev 4, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
    |__ Port 7: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 8: Dev 6, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 8: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M

Bus 001 Device 003: ID 0bda:b812 Realtek Semiconductor Corp. 802.11ac NIC
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xb812 
  bcdDevice            2.10
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0

zuzia-dev, 2019-11-30:
Detailed logs:

- [lsusb](https://pastebin.com/q9Ewg4hH)

- [dmesg](https://pastebin.com/aBKBDRM9)

wonko6x9, 2019-11-30:
> [dmsg](https://pastebin.com/pUYMhxX6)

> [lusb -v](https://pastebin.com/BQLLS9Uw)

gnumoksha, 2020-01-10:
I'm experiencing the same as @zuzia-dev

yufalcon, 2020-10-11:
> Port USB 3.0.
>
> ```
> modprobe 88x2bu rtw_switch_usb_mode=1
> lsusb -t
> /:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
>     |__ Port 5: Dev 2, If 0, Class=Vendor Specific Class, [...]

maybe its fake port from this product