cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #28: Unable to make it work on current Ubuntu 19.04 kernel 5.0.x

Polve, 2019-07-02:
The module loads and the dongle is found but it seems like it's unable to authenticate and tries forever, requesting password again and again

rrafal1337, 2019-07-09:
I can confirm it works correctly both on 5.0.x and 5.1.x kernels on Fedora 29 and Fedora 30

fspoettel, 2019-07-11:
Facing the same issue, `Ubuntu 19.04` on kernel `5.0.0-20-generic`.

ruabmbua, 2019-07-23:
Facing the same issue, arch linux kernel 5.2.2-arch1-1-ARCH

Niproblema, 2019-07-23:
Can someone pin `dmesg` error on usb insertion or on auth?

jonahisadev, 2019-08-01:
I'm having the same issue on Ubuntu 18.04 kernel 4.18.0-25-generic. I'm attempting to upgrade my distribution, but I would imagine it's gonna give me the same results

ruabmbua, 2019-08-03:
I tested a little bit more, and it seems like it only fails on USB 3. If I plug it into

jonahisadev, 2019-08-03:
@ruabmbua I totally thought I had tried this before without it working, but it appears to be working now. Thank you!

lvshow2001, 2020-04-01:
is that issue same with mine???
I am sure the ssid and password are match,but it always like this:
```
root@pi-PC:/home/pi# sudo wpa_supplicant -c/etc/wpa_supplicant.conf  -iwlx1cbfceb98413
Successfully initialized wpa_supplicant
wlx1cbfceb98413: Trying to associate with fc:d7:33:ba:e5:e0 (SSID='tcm' freq=2412 MHz)
wlx1cbfceb98413: CTRL-EVENT-ASSOC-REJECT status_code=1
wlx1cbfceb98413: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlx1cbfceb98413: Trying to associate with fc:d7:33:ba:e5:e0 (SSID='tcm' freq=2412 MHz)
wlx1cbfceb98413: Authentication with fc:d7:33:ba:e5:e0 timed out.
wlx1cbfceb98413: CTRL-EVENT-DISCONNECTED bssid=fc:d7:33:ba:e5:e0 reason=3 locally_generated=1
wlx1cbfceb98413: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlx1cbfceb98413: Trying to associate with fc:d7:33:ba:e5:e0 (SSID='tcm' freq=2412 MHz)
wlx1cbfceb98413: CTRL-EVENT-ASSOC-REJECT status_code=1
wlx1cbfceb98413: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="tcm" auth_failures=1 duration=10 reason=CONN_FAILED
wlx1cbfceb98413: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlx1cbfceb98413: CTRL-EVENT-SSID-REENABLED id=0 ssid="tcm"
wlx1cbfceb98413: Trying to associate with fc:d7:33:ba:e5:e0 (SSID='tcm' freq=2412 MHz)
wlx1cbfceb98413: CTRL-EVENT-ASSOC-REJECT status_code=1
wlx1cbfceb98413: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="tcm" auth_failures=2 duration=23 reason=CONN_FAILED
wlx1cbfceb98413: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlx1cbfceb98413: CTRL-EVENT-SSID-REENABLED id=0 ssid="tcm"
wlx1cbfceb98413: Trying to associate with fc:d7:33:ba:e5:e0 (SSID='tcm' freq=2412 MHz)
wlx1cbfceb98413: Associated with fc:d7:33:ba:e5:e0
wlx1cbfceb98413: CTRL-EVENT-DISCONNECTED bssid=fc:d7:33:ba:e5:e0 reason=0 locally_generated=1
wlx1cbfceb98413: WPA: 4-Way Handshake failed - pre-shared key may be incorrect
wlx1cbfceb98413: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="tcm" auth_failures=3 duration=46 reason=WRONG_KEY
wlx1cbfceb98413: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="tcm" auth_failures=4 duration=77 reason=CONN_FAILED
wlx1cbfceb98413: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=CN
wlx1cbfceb98413: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD

```
my linux version info as below:
```
root@pi-PC:/home/pi# uname -a
Linux pi-PC 4.15.0-30deepin-generic #31 SMP Fri Nov 30 04:29:02 UTC 2018 x86_64 GNU/Linux
```
more info
```
oot@pi-PC:/home/pi# nmcli
ens33: disconnected
        "Intel 82545EM Gigabit Ethernet Controller (Copper) (PRO/1000 MT Single Port Adapter)"
        1 connection available
        ethernet (e1000), 00:0C:29:EA:6D:C8, autoconnect, hw, mtu 1500

wlx1cbfceb98413: disconnected
        "Realtek 802.11ac NIC"
        2 connections available
        wifi (rtl88x2bu), 1C:BF:CE:B9:84:13, hw, mtu 1500

lo: unmanaged
        "lo"
        loopback (unknown), 00:00:00:00:00:00, sw, mtu 65536
```
```
root@pi-PC:/home/pi# dkms status
bcmwl, 6.30.223.271+bdcom, 4.15.0-30deepin-generic, x86_64: installed
deepin-anything, 0.0, 4.15.0-30deepin-generic, x86_64: installed
huawei-wmi, 1.0, 4.15.0-30deepin-generic, x86_64: installed
mincores, 0.2.0, 4.15.0-30deepin-generic, x86_64: installed
rtl88x2bu, 5.3.1, 4.15.0-30deepin-generic, x86_64: installed
```