cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #31: Unable to connect to 5.8 GHz networks with 0bda:b812 Realtek USB adapter on 18.04 with 4.15.0-54-generic x86_64

aviogit, 2019-07-20:
Hi @cilynx,

first, thanks for the great work, I've read that your driver helped dozen of people make these adapters work.

I bought this cheap adapter: https://www.amazon.it/VGROUND-802-11ac-Adattatore-Wireless-Supporta/dp/B07D9HXRZ6

Your driver almost works with this adapter, it brings up a wireless interface and it successfully connected to a 2.4 GHz (even with a few retries).

> wlxbcec23c36ff0  unassociated  Nickname:"<WIFI@REALTEK>"
>           Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   
>           Sensitivity:0/0  
>           Retry:off   RTS thr:off   Fragment thr:off
>           Power Management:off
>           Link Quality:0  Signal level:0  Noise level:0
>           Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
>           Tx excessive retries:0  Invalid misc:0   Missed beacon:0

The problem arises with 5.8 GHz networks. I've tried both this version and the old v5.2.4.4 (below you can find the dmesg log for both for v5.3.1 and v5.2.4.4) but the adapter never connects. Any ideas about what I could do to make it work?

**v5.3.1**
> [Sat Jul 20 13:15:16 2019] ---------------------------------------------
> [Sat Jul 20 13:15:21 2019] RTW: module init start
> [Sat Jul 20 13:15:21 2019] RTW: rtl88x2bu v5.3.1_27678.20180430_COEX20180427-5959
> [Sat Jul 20 13:15:21 2019] RTW: build time: Jul 20 2019 13:14:52
> [Sat Jul 20 13:15:21 2019] RTW: rtl88x2bu BT-Coex version = COEX20180427-5959
> [Sat Jul 20 13:15:21 2019] RTW: rtw_inetaddr_notifier_register
> [Sat Jul 20 13:15:21 2019] usbcore: registered new interface driver rtl88x2bu
> [Sat Jul 20 13:15:21 2019] RTW: module init ret=0
> [Sat Jul 20 13:15:27 2019] usb 1-2.3: new high-speed USB device number 19 using xhci_hcd
> [Sat Jul 20 13:15:27 2019] usb 1-2.3: New USB device found, idVendor=0bda, idProduct=b812
> [Sat Jul 20 13:15:27 2019] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [Sat Jul 20 13:15:27 2019] usb 1-2.3: Product: USB3.0 802.11ac 1200M Adapter
> [Sat Jul 20 13:15:27 2019] usb 1-2.3: Manufacturer: Realtek
> [Sat Jul 20 13:15:27 2019] usb 1-2.3: SerialNumber: 123456
> [Sat Jul 20 13:15:28 2019] RTW: 
>                            usb_endpoint_descriptor(0):
> [Sat Jul 20 13:15:28 2019] RTW: bLength=7
> [Sat Jul 20 13:15:28 2019] RTW: bDescriptorType=5
> [Sat Jul 20 13:15:28 2019] RTW: bEndpointAddress=84
> [Sat Jul 20 13:15:28 2019] RTW: wMaxPacketSize=512
> [Sat Jul 20 13:15:28 2019] RTW: bInterval=0
> [Sat Jul 20 13:15:28 2019] RTW: RT_usb_endpoint_is_bulk_in = 4
> [Sat Jul 20 13:15:28 2019] RTW: 
>                            usb_endpoint_descriptor(1):
> [Sat Jul 20 13:15:28 2019] RTW: bLength=7
> [Sat Jul 20 13:15:28 2019] RTW: bDescriptorType=5
> [Sat Jul 20 13:15:28 2019] RTW: bEndpointAddress=5
> [Sat Jul 20 13:15:28 2019] RTW: wMaxPacketSize=512
> [Sat Jul 20 13:15:28 2019] RTW: bInterval=0
> [Sat Jul 20 13:15:28 2019] RTW: RT_usb_endpoint_is_bulk_out = 5
> [Sat Jul 20 13:15:28 2019] RTW: 
>                            usb_endpoint_descriptor(2):
> [Sat Jul 20 13:15:28 2019] RTW: bLength=7
> [Sat Jul 20 13:15:28 2019] RTW: bDescriptorType=5
> [Sat Jul 20 13:15:28 2019] RTW: bEndpointAddress=6
> [Sat Jul 20 13:15:28 2019] RTW: wMaxPacketSize=512
> [Sat Jul 20 13:15:28 2019] RTW: bInterval=0
> [Sat Jul 20 13:15:28 2019] RTW: RT_usb_endpoint_is_bulk_out = 6
> [Sat Jul 20 13:15:28 2019] RTW: 
>                            usb_endpoint_descriptor(3):
> [Sat Jul 20 13:15:28 2019] RTW: bLength=7
> [Sat Jul 20 13:15:28 2019] RTW: bDescriptorType=5
> [Sat Jul 20 13:15:28 2019] RTW: bEndpointAddress=87
> [Sat Jul 20 13:15:28 2019] RTW: wMaxPacketSize=64
> [Sat Jul 20 13:15:28 2019] RTW: bInterval=3
> [Sat Jul 20 13:15:28 2019] RTW: RT_usb_endpoint_is_int_in = 7, Interval = 3
> [Sat Jul 20 13:15:28 2019] RTW: 
>                            usb_endpoint_descriptor(4):
> [Sat Jul 20 13:15:28 2019] RTW: bLength=7
> [Sat Jul 20 13:15:28 2019] RTW: bDescriptorType=5
[... 1933 lines cut ...]
> [Sat Jul 20 12:47:08 2019] RTW: ==>rtw_ps_processor .fw_state(8)
> [Sat Jul 20 12:47:10 2019] RTW: cfg80211_rtw_scan(wlxbcec23c36ff0)
> [Sat Jul 20 12:47:18 2019] RTW: rtw_scan_timeout_handler(wlxbcec23c36ff0) fw_state=808
> [Sat Jul 20 12:47:18 2019] RTW: rtw_indicate_scan_done(wlxbcec23c36ff0)
> [Sat Jul 20 12:47:18 2019] RTW: cfg80211_rtw_disconnect(wlxbcec23c36ff0) - Start to Disconnect
> [Sat Jul 20 12:47:19 2019] RTW: survey done event(d0) band:0 for wlxbcec23c36ff0
> [Sat Jul 20 12:47:19 2019] RTW: hw_var_set_sec_dk_cfg: [HW_VAR_SEC_DK_CFG] 0x680=0x0000030c
> [Sat Jul 20 12:47:19 2019] RTW: cfg80211_rtw_disconnect...call rtw_indicate_disconnect
> [Sat Jul 20 12:47:19 2019] RTW: rtw_free_assoc_resources-wlxbcec23c36ff0 tgt_network MacAddress=bc:cf:4f:08:56:8d ssid=TechnoViking-5G
> [Sat Jul 20 12:47:19 2019] RTW: free disconnecting network of scanned_queue
> [Sat Jul 20 12:47:19 2019] RTW: rtw_cfg80211_unlink_bss(): cfg80211_unlink TechnoViking-5G!!
> [Sat Jul 20 12:47:19 2019] RTW: rtw_cfg80211_indicate_disconnect(wlxbcec23c36ff0)
> [Sat Jul 20 12:47:19 2019] RTW: cfg80211_rtw_disconnect(wlxbcec23c36ff0) return 0
> [Sat Jul 20 12:47:19 2019] RTW: [HW_VAR_CHECK_TXBUF] Empty in 16 ms
> [Sat Jul 20 12:47:20 2019] RTW: rtw_surveydone_event_callback(wlxbcec23c36ff0) fw_state:0x8
> [Sat Jul 20 12:47:20 2019] RTW: rtw_indicate_scan_done(wlxbcec23c36ff0)
> [Sat Jul 20 12:47:20 2019] RTW: rtw_reset_securitypriv(wlxbcec23c36ff0) - End to Disconnect
> [Sat Jul 20 12:47:20 2019] RTW: rtw_ndev_notifier_call(wlxbcec23c36ff0) state:4
> [Sat Jul 20 12:47:20 2019] IPv6: ADDRCONF(NETDEV_UP): wlxbcec23c36ff0: link is not ready
> [Sat Jul 20 12:47:20 2019] RTW: cfg80211_rtw_get_txpower
> [Sat Jul 20 12:47:20 2019] RTW: cfg80211_rtw_scan(wlxbcec23c36ff0)
> [Sat Jul 20 12:47:25 2019] ---------------------------------------------
> [Sat Jul 20 12:47:29 2019] RTW: rtw_scan_timeout_handler(wlxbcec23c36ff0) fw_state=808
> [Sat Jul 20 12:47:29 2019] RTW: rtw_indicate_scan_done(wlxbcec23c36ff0)
> [Sat Jul 20 12:47:29 2019] RTW: cfg80211_rtw_scan(wlxbcec23c36ff0)

aviogit, 2019-07-20:
My god, I've tried the same adapter on a laptop with 16.04 and @cilynx driver, and the speedtest often reaches 150 Mbps in down and 100 Mbps in up. I've also tested the Win10 built-in driver and it's crappy as the whole Windows ecosystem (it reaches a maximum of 30 Mbps both in up and down).

Morale: the adapter is working, my machine is the problem.

aviogit, 2019-07-20:
Ok, as always happens, the human is the actual problem. As soon as I removed the WWI USB cable that I was using to test the adapter, the driver established the connection. The speedtest now looks like this:

[image]

Also the timings were suggesting that something was wrong...

> --- working.txt		2019-07-20 16:30:19.413504606 +0200
> +++ not-working.txt	2019-07-20 16:30:57.505815004 +0200
> @@ -1,9 +1,8 @@
> -usb 2-2: new high-speed USB device number 2 using xhci_hcd
> -usb 2-2: New USB device found, idVendor=0bda, idProduct=b812
> -usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> -usb 2-2: Product: USB3.0 802.11ac 1200M Adapter
> -usb 2-2: Manufacturer: Realtek
> -usb 2-2: SerialNumber: 123456
> +RTW: module init start
> +RTW: rtl88x2bu v5.3.1_27678.20180430_COEX20180427-5959
> +RTW: build time: Jul 20 2019 16:24:41
> +RTW: rtl88x2bu BT-Coex version = COEX20180427-5959
> +RTW: rtw_inetaddr_notifier_register
>  RTW: 
>  usb_endpoint_descriptor(0):
>  RTW: bLength=7
> @@ -143,9 +142,9 @@
>  RTW: rtl8822b_fw_dl Download Firmware from array success
>  RTW: NIC FW Version:22 SubVersion:6 FW size:145104
>  RTW: SetHwReg: bMacPwrCtrlOn=0
> -RTW: hal_read_mac_hidden_rpt OK! (1, 36ms), fwdl:1, id:0x19
> +RTW: hal_read_mac_hidden_rpt OK! (1, 276ms), fwdl:1, id:0x19
>  RTW: EEPROM Disable BT-coex by hal_spec
> -RTW: rtw_hal_read_chip_info in 1920 ms
> +RTW: rtw_hal_read_chip_info in 15088 ms
>  RTW: init_channel_set((null)) ChannelPlan ID:0x7f, ch num:37
>  RTW: NR_RECVBUFF: 8
>  RTW: MAX_RECVBUF_SZ: 32768
> @@ -159,15 +158,18 @@
>  RTW: can't get autopm:
>  RTW: rtw_macaddr_cfg mac addr:bc:ec:23:c3:6f:f0
>  RTW: bDriverStopped:True, bSurpriseRemoved:False, bup:0, hw_init_completed:0
> -RTW: rtw_wiphy_alloc(phy1)
> -RTW: rtw_wdev_alloc(padapter=ffffc90001bf8000)
> -RTW: rtw_wiphy_register(phy1)
> +RTW: rtw_wiphy_alloc(phy2)
> +RTW: rtw_wdev_alloc(padapter=00000000a8cf3123)
> +RTW: rtw_wiphy_register(phy2)
>  RTW: Register RTW cfg80211 vendor cmd(0x67) interface
> -RTW: rtw_ndev_init(wlan1) if1 mac_addr=bc:ec:23:c3:6f:f0
> -RTW: rtw_ndev_notifier_call(wlan1) state:16
> -RTW: rtw_ndev_notifier_call(wlan1) state:5
> +RTW: rtw_ndev_init(wlan0) if1 mac_addr=bc:ec:23:c3:6f:f0
> +RTW: rtw_ndev_notifier_call(wlan0) state:16
>  RTW: cfg80211_rtw_get_txpower
> -rtl88x2bu 2-2:1.0 wlxbcec23c36ff0: renamed from wlan1
> +RTW: rtw_ndev_notifier_call(wlan0) state:5
> +usbcore: registered new interface driver rtl88x2bu
> +RTW: module init ret=0
> +RTW: cfg80211_rtw_get_txpower
> +rtl88x2bu 1-2:1.0 wlxbcec23c36ff0: renamed from wlan0
>  RTW: rtw_ndev_notifier_call(wlxbcec23c36ff0) state:10
>  IPv6: ADDRCONF(NETDEV_UP): wlxbcec23c36ff0: link is not ready
>  RTW: rtw_ndev_notifier_call(wlxbcec23c36ff0) state:13
> @@ -189,7 +191,7 @@
>  RTW: is_valid_id_status: HALMAC_FEATURE_DUMP_PHYSICAL_EFUSE
>  RTW: is_valid_id_status: HALMAC_FEATURE_DUMP_PHYSICAL_EFUSE
>  RTW: is_valid_id_status: HALMAC_FEATURE_DUMP_PHYSICAL_EFUSE
> -RTW: rtl8822bu_init in 2012ms
> +RTW: rtl8822bu_init in 12692ms
>  RTW: rtw_hal_set_macaddr_port wlxbcec23c36ff0- hw port(0) mac_addr =bc:ec:23:c3:6f:f0
>  RTW: rtw_hal_get_macaddr_port wlxbcec23c36ff0- hw port(0) mac_addr =bc:ec:23:c3:6f:f0
>  RTW: wlxbcec23c36ff0- hw port(0) mac_addr =bc:ec:23:c3:6f:f0

[not-working.txt](https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/files/3413713/not-working.txt)
[working.txt](https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/files/3413714/working.txt)

aviogit, 2019-07-20:
Last image, now I stop it, I promise :)

[image]

cilynx, 2019-07-20:
Glad you got it working.  I've run into similar issues on my test bench using a cheap USB extension cable.

aviogit, 2019-07-20:
No pain, no gain :joy:
Thank you again for the great driver!

aviogit, 2019-07-27:
I think I spoke too early. It connects quite rarely, and after many many retries, especially on the 5 GHz network.

aviogit, 2019-10-19:
Now it works perfectly, since a few weeks it connects at every boot in seconds (current kernel is 4.15.0-65-generic on Ubuntu 18.04.3). I'd mark this issue as solved. Thanks again for the excellent driver!

closed 2019-10-19 by aviogit