cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #33: TP-Link Archer T4U AC1300 V3 - KALI on Raspberry 3 B

joridos, 2019-07-28:
I'm trying to compile this drive in the Kali on a Raspberry 3 Model B.

LOG:
```bash
root@kali:/usr/src/rtl88x2bu-5.3.1# make 
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.55-Re4son-v8+/build M=/usr/src/rtl88x2bu-5.3.1  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.55-Re4son-v8+'
  CC [M]  /usr/src/rtl88x2bu-5.3.1/core/rtw_cmd.o
gcc: error: unrecognized argument in option ‘-mabi=apcs-gnu’
gcc: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
gcc: error: unrecognized command line option ‘-mapcs’; did you mean ‘--specs’?
gcc: error: unrecognized command line option ‘-mno-sched-prolog’; did you mean ‘-Wno-sign-promo’?
gcc: error: unrecognized command line option ‘-msoft-float’
make[2]: *** [scripts/Makefile.build:304: /usr/src/rtl88x2bu-5.3.1/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1517: _module_/usr/src/rtl88x2bu-5.3.1] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.55-Re4son-v8+'
make: *** [Makefile:2011: modules] Error 2
```

Additional information:
```bash
root@kali:/usr/src/rtl88x2bu-5.3.1# uname -a
Linux kali 4.19.55-Re4son-v8+ #1 SMP PREEMPT Sun Jun 30 10:40:44 UTC 2019 aarch64 GNU/Linux
root@kali:/usr/src/rtl88x2bu-5.3.1# lsb_release -a
No LSB modules are available.
Distributor ID:	Kali
Description:	Kali GNU/Linux Rolling
Release:	2019.2
Codename:	n/a
```

mbwmbw1337, 2019-09-03:
Any updates on this issue? I'm having the same problem.

joridos, 2019-09-03:
i'm still having the issue.

mbwmbw1337, 2019-09-03:
>i'm still having the issue.

@joridos I figured it out sorry for not updating here. Use:
`make ARCH="arm64"` instead of `make`

Let me know if this works for you.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>