cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #37: Bad return status for module build on kernel: 4.19.66-v7l+ (armv7l)

FattyboyN, 2019-09-04:
DKMS make.log for rtl88x2bu-5.3.1 for kernel 4.19.66-v7l+ (armv7l)
Wed  4 Sep 13:36:30 AEST 2019
/bin/sh: 1: bc: not found
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.19.66-v7l+/build M=/var/lib/dkms/rtl88x2bu/5.3.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.66-v7l+'
Makefile:614: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.19.66-v7l+'
make: *** [Makefile:2011: modules] Error 2

How to fix this?

kaydubbed, 2019-09-04:
Pi4?

You need to change the makefile.

sed -i 's/I386_PC = y/I386_PC = n/' Makefile
sed -i 's/ARM_RPI = n/ARM_RPI = y/' Makefile

cilynx, 2019-09-22:
As @kaydubbed says, you need to make sure you're [enabling the correct platform in the Makefile](https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/blob/master/Makefile#L100-L101).

Also, you're missing `bc` per `/bin/sh: 1: bc: not found`.  You'll need to install `bc` for the [gcc version check in the Makefile](https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/blob/master/Makefile#L19).  

Can you confirm what system you're compiling for and if you have `expr` available by default?  We could potentially replace `bc` with `expr`, but there's not much point if it's not available either.