cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #39: AC mode AP?

dmarkey, 2019-09-06:
Trying to get hostapd to work with this USB dongle.

Configuring for AC but negotiation seems to only go N.

Anyone got an AC access point to work with this chipset?

mehditlili, 2019-09-17:
Here is my config
```
interface=wlx****
driver=nl80211

ssid=*****
hw_mode=a
channel=44
country_code=DE
ignore_broadcast_ssid=0

wpa=2
wpa_passphrase=***
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
auth_algs=1
macaddr_acl=0
```

powderluv, 2020-01-29:
Were you able to get it to .ac ?

The best I can get to is n
wlp1s0u1u1  IEEE 802.11an  ESSID:"borg"  Nickname:"<WIFI@REALTEK>"
          Mode:Master  Frequency:5.22 GHz  Access Point: 3A:5D:33:AA:0E:3D   
          Bit Rate:144.4 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

Using:

interface=wlp1s0u1u1
driver=nl80211
country_code=US
#ieee80211d=1
#ieee80211h=1
ieee80211n=1          # 802.11n support
ieee80211ac=1         # 802.11ac support
wmm_enabled=1         # QoS support
ssid=borg
hw_mode=a
channel=44
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=****
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

Using @mehditlili config above I can only get 54g speeds and not even n

powderluv, 2020-01-29:
btw you may have to use a custom hostap and rtl driver instead of nl80211
https://aur.archlinux.org/packages/hostapd-rtl871xdrv/

yufalcon, 2020-10-11:
> Trying to get hostapd to work with this USB dongle.
> 
> Configuring for AC but negotiation seems to only go N.
> 
> Anyone got an AC access point to work with this chipset?

whats your tool to make a hostspot?