cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #41: Not working with newly released Kernel 5.3.0 on ubuntu 18.04

mpdt, 2019-09-20:
Hi Cilynx,

Even though the module already exists in the DKMS tree, when inserting the WiFi dongle it is not recognised at all. Modprobing again makes no difference.

As a result I have reverted back to 5.2.16, where it is working brilliantly :-)

CGarces, 2019-09-20:
Hi  @mpdt and @cilynx 
I'm not sure if it's related, please test this patch
https://github.com/Mange/rtl8192eu-linux-driver/commit/e7361e951c15d998f8c63b45478ef320d447f769

mpdt, 2019-09-20:
Hi @CGarces 
Looking at your patch here, it does look like it could be the same issue.  I don't have time to test it myself right now, but when I do I will.  Thanks for your help :-)

closed 2019-09-22 by cilynx

cilynx, 2019-09-22:
@karubabu's fix looks good to me -- merged.  Could someone please confirm functionality and either provide a `uname -a` and `gcc --version` here or send a PR to update the [README](https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/blob/master/README.md)?  Thanks!

AnonymFegis, 2019-09-22:
Works on a pre-release 19.10 Ubuntu machine (proposed enabled)

fredrik@Q4:~$ uname -a
Linux Q4 5.3.0-12-generic #13-Ubuntu SMP Tue Sep 17 12:35:50 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

fredrik@Q4:~$ gcc --version
gcc (Ubuntu 9.2.1-8ubuntu1) 9.2.1 20190909
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

cilynx, 2019-09-22:
In case y'all bleeding-edge folks are interested, I stumbled upon the 5.6.1 upstream version a couple days ago and have now ported the updates.  I'm getting tired of repo-roulette, so the below will be the last new `rtl88x2bu` repo and new upstream versions will be managed in branches going forward.  Once I get a few confirmations that 5.6.1 is working as expected, I'll archive this repo off like the older ones and add a link in the README to our new home.

https://github.com/cilynx/rtl88x2bu

PR #42 (2019-09-22): add support kernel-5.3
fix #41 
Fetch from https://github.com/jwrdegoede/rtl8189ES_linux/commit/50da9756282edf5008a5a567275b33c1d2f16197