cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #44: raspberry 4b build error

fjson, 2019-10-06:
dkms build -m rtl88x2bu -v ${VER}

Error! echo
Your kernel headers for kernel 4.19.75-v7l+ cannot be found at
/lib/modules/4.19.75-v7l+/build or /lib/modules/4.19.75-v7l+/source.

rrafal1337, 2019-10-06:
Hi. You need to install package kernel-headers as well. Version of this package must fit your version of kernel. Usually if you use default kernel then just install **kernel-headers** and **bc** and **gcc** and **dkms** as well.

fjson, 2019-10-06:
> Hi. You need to install package kernel-headers as well. Version of this package must fit your version of kernel. Usually if you use default kernel then just install **kernel-headers** and **bc** and **gcc** and **dkms** as well.

raspberrypi-kernel-headers  1.20190819-1
uname -r  4.19.75-v7l+

fjson, 2019-10-06:
Is it that my kernel and raspberrypi-kernel-headers versions are not uniform, but they are all up to date.

rrafal1337, 2019-10-06:
/lib/modules/4.19.75-v7l+/build should be a symlink to your kernel sources directory. For me this link points to:

**ls -l /lib/modules/5.2.17-200.fc30.armv7hl/build**
lrwxrwxrwx. 1 root root 40 09-23 22:20 /lib/modules/5.2.17-200.fc30.armv7hl/build -> /usr/src/kernels/5.2.17-200.fc30.armv7hl

lexdevelop, 2019-12-26:
I have same issue:
ls -l /lib/modules/4.19.55-6-osmc/build
`lrwxrwxrwx 1 root root 36 Dec 26 20:17 /lib/modules/4.19.55-6-osmc/build -> /usr/src/rbp2-headers-4.19.55-6-osmc`

This is build log:
cat /var/lib/dkms/rtl88x2bu/5.3.1/build/make.log
DKMS make.log for rtl88x2bu-5.3.1 for kernel 4.19.55-6-osmc (armv7l)
Thu Dec 26 23:08:25 CET 2019
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.55-6-osmc/build M=/var/lib/dkms/rtl88x2bu/5.3.1/build  modules
make[1]: Entering directory '/usr/src/rbp2-headers-4.19.55-6-osmc'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o
In file included from ./include/linux/compat.h:16:0,
                 from ./include/linux/ethtool.h:17,
                 from ./include/linux/netdevice.h:41,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service_linux.h:30,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_debug.c:17:
./include/linux/if.h:28:54: fatal error: sys/socket.h: No such file or directory
 #include <sys/socket.h>   /* for struct sockaddr.  */
                                                      ^
compilation terminated.
In file included from ./include/linux/compat.h:16:0,
                 from ./include/linux/ethtool.h:17,
                 from ./include/linux/netdevice.h:41,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service_linux.h:30,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_cmd.c:17:
./include/linux/if.h:28:54: fatal error: sys/socket.h: No such file or directory
 #include <sys/socket.h>   /* for struct sockaddr.  */
                                                      ^
compilation terminated.
scripts/Makefile.build:303: recipe for target '/var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o' failed
make[2]: *** [/var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o] Error 1
In file included from ./include/linux/compat.h:16:0,
                 from ./include/linux/ethtool.h:17,
                 from ./include/linux/netdevice.h:41,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service_linux.h:30,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_io.c:47:
./include/linux/if.h:28:54: fatal error: sys/socket.h: No such file or directory
 #include <sys/socket.h>   /* for struct sockaddr.  */
                                                      ^
make[2]: *** Waiting for unfinished jobs....
compilation terminated.
scripts/Makefile.build:303: recipe for target '/var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o' failed
make[2]: *** [/var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o] Error 1
scripts/Makefile.build:303: recipe for target '/var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o' failed
make[2]: *** [/var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o] Error 1
In file included from ./include/linux/compat.h:16:0,
                 from ./include/linux/ethtool.h:17,
                 from ./include/linux/netdevice.h:41,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service_linux.h:30,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_security.c:17:
./include/linux/if.h:28:54: fatal error: sys/socket.h: No such file or directory
 #include <sys/socket.h>   /* for struct sockaddr.  */
                                                      ^
compilation terminated.
scripts/Makefile.build:303: recipe for target '/var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_security.o' failed
make[2]: *** [/var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_security.o] Error 1
Makefile:1517: recipe for target '_module_/var/lib/dkms/rtl88x2bu/5.3.1/build' failed
make[1]: *** [_module_/var/lib/dkms/rtl88x2bu/5.3.1/build] Error 2
make[1]: Leaving directory '/usr/src/rbp2-headers-4.19.55-6-osmc'
Makefile:2011: recipe for target 'modules' failed
make: *** [modules] Error 2

Any idea?

rrafal1337, 2019-12-27:
@lexdevelop you need probably just to install **glibc-headers** package and retry compilation again.

> fatal error: sys/socket.h: No such file or directory

lexdevelop, 2019-12-27:
@rrafal1337 I manage to build this on Raspbian, it looks like the issue is about OSMC.
Here is a similar issue:
https://discourse.osmc.tv/t/has-anybody-tried-to-install-wireguard/73506/39

rbp2-headers has some issue with **headers_install.sh**
```
echo "Usage: headers_install.sh OUTDIR SRCDIR [FILES...]"
echo
echo "Prepares kernel header files for use by user space, by removing"
echo "all compiler.h definitions and #includes, removing any"
echo "#ifdef __KERNEL__ sections, and putting __underscores__ around"
echo "asm/inline/volatile keywords."
echo
echo "OUTDIR: directory to write each userspace header FILE to."
echo "SRCDIR: source directory where files are picked."
echo "FILES:  list of header files to operate on."
```

This means if we have:
```
#ifndef __KERNEL__ 
#include <sys/socket.h>     /* for struct sockaddr. */ 
#endif
```

It will not look at the user space.

Have no idea how to solve this, since I need this driver on [OSMC](https://osmc.tv/)

rrafal1337, 2019-12-27:
@lexdevelop Checked already on two Linux disributions (Arch Linux and Fedora 31) and it seems this file in both cases is in **/usr/include/sys/socket.h**
if you have those files somewhere else then maybe issue can be solved by copying files to right place or make symlinks.