cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #48: Driver doesn't work on kernel 5.3.6 - Fedora 30

rrafal1337, 2019-10-22:
Hi. Driver previously worked with 5.2.18. I was able to compile driver on 5.3.x but when it starts there is error in dmesg about module:
https://pastebin.com/7Vaw3dJJ

rrafal1337, 2019-10-22:
Sister project (rtl8821cu) already have solution implemented.
Details see here: https://github.com/brektrou/rtl8821CU/issues/15#issuecomment-545037120

cilynx, 2019-10-23:
The vendor command policy fix mentioned in that thread was merged here from #42.  We'll need to do some digging to find what's failing with the newer kernel.

silvernode, 2019-10-24:
@cilynx Just wanted to confirm that I have no issues on Fedora 31 with kernel 5.3.6.  

`$ Linux 5.3.6-300.fc31.x86_64`  

I will be happy to provide useful information if requested.

fleamour, 2019-11-12:
Fedora GNOME 31:
```
Error! echo
Your kernel headers for kernel 5.3.7-301.fc31.x86_64 cannot be found at
/lib/modules/5.3.7-301.fc31.x86_64/build or /lib/modules/5.3.7-301.fc31.x86_64/source.
```

rrafal1337, 2019-11-12:
> Fedora GNOME 31:
> 
> ```
> Error! echo
> Your kernel headers for kernel 5.3.7-301.fc31.x86_64 cannot be found at
> /lib/modules/5.3.7-301.fc31.x86_64/build or /lib/modules/5.3.7-301.fc31.x86_64/source.
> ```

Hi. You need to install dkms, kernel-devel, kernel-headers, gcc, bc packages.

I can confirm driver works for me with latest kernel on Fedora 30 - 5.3.8-200.fc30.x86_64

Thank you for support.

closed 2019-11-12 by rrafal1337