cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #57: Exec format error when building RPi4

antoinezambelli, 2020-01-25:
Not sure where this comes from, I followed the README instructions on a fresh Buster build. It fails pretty quickly on the `build` step, here is my log file. Any ideas?

```
DKMS make.log for rtl88x2bu-5.3.1 for kernel 4.19.75-v7l+ (armv7l)
Sat 25 Jan 2020 10:36:54 AM PST
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.75-v7l+/build M=/var/lib/dkms/rtl88x2bu/5.3.$
make[1]: Entering directory '/usr/src/linux-headers-4.19.75-v7l+'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o] E$
make[2]: *** Waiting for unfinished jobs....
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_securit$
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o] $
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o$
make[1]: *** [Makefile:1519: _module_/var/lib/dkms/rtl88x2bu/5.3.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.75-v7l+'
make: *** [Makefile:2011: modules] Error 2

```

powderluv, 2020-01-29:
Did you do:

# Configure for RasPi
sed -i 's/I386_PC = y/I386_PC = n/' Makefile
sed -i 's/ARM_RPI = n/ARM_RPI = y/' Makefile

powderluv, 2020-01-29:
I have it running well on my RPi4 but I am using the 64bit Gentoo build

antoinezambelli, 2020-01-31:
Yeah I edited the Makefile. This error looks very similar to a mismatched kernel headers error I just learned about (yay?). I'll dig into it over the weekend and confirm. If it is the kernel mismatch then it's not an issue with this driver but with my Pi that I would need to reboot to the correct kernel/header version.

lvshow2001, 2020-03-31:
when I run 'sudo dkms build -m rtl88x2bu -v ${VER}' I got a error
my error log:
DKMS make.log for rtl88x2bu-5.3.1 for kernel 4.19.75-v7+ (armv7l)
Tue 31 Mar 2020 05:17:21 PM CST
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.75-v7+/build M=/var/lib/dkms/rtl88x2bu/5.3.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.75-v7+'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o] Error 2
make[2]: *** Waiting for unfinished jobs....
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_security.o] Error 2
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o] Error 2
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o] Error 2
make[1]: *** [Makefile:1519: _module_/var/lib/dkms/rtl88x2bu/5.3.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.75-v7+'

my raspbian build:
Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux

How to fix it or avoid it ? thanks!

powderluv, 2020-03-31:
you seem to be building for 5.3.1 and have 4.19 installed.