cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #62: modprobe: FATAL: Module 88x2bu not found in directory /lib/modules/5.4.0-kali3-amd64

A2hari, 2020-02-16:
unable to install the driver on the kali linux 2020.1 
 uname -a
Linux swetha 5.4.0-kali3-amd64 #1 SMP Debian 5.4.13-1kali1 (2020-01-20) x86_64 GNU/Linux

mtnsec, 2020-02-21:
Same here on Kali 2020.1 on Raspberry Pi:
```
Error! Bad return status for module build on kernel: 4.19.93-Re4son-v8l+ (aarch64)
Consult /var/lib/dkms/rtl88x2bu/5.3.1/build/make.log for more information.
```

A2hari, 2020-02-22:
I got this issue resolved using the following commands try theese

apt update
apt install bc -y
git clone https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959.git
cd rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959
VER=$(sed -n 's/\PACKAGE_VERSION="\(.*\)"/\1/p' dkms.conf)
sudo rsync -rvhP ./ /usr/src/rtl88x2bu-${VER}
sudo dkms add -m rtl88x2bu -v ${VER}
sudo dkms build -m rtl88x2bu -v ${VER}
sudo dkms install -m rtl88x2bu -v ${VER}
sudo modprobe 88x2bu

[here ](https://forums.kali.org/showthread.php?44810-Realtek-RTL8812BU-Driver-for-Kali-Linux)is the official documentation support for rlt99x2bu for kali linux hop this works

closed 2020-02-22 by A2hari