cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #67: Error! Bad return status for module build on kernel: 5.3.0-kali3-686-pae (i686)

roy668899, 2020-04-10:
**bash** 
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j4 KERNELRELEASE=5.3.0-kali3-686-pae KVER=5.3.0-kali3-686-pae src=/usr/src...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.3.0-kali3-686-pae (i686)
/build/make.log for more information.

**The content of make.log:**

DKMS make.log for rtl88x2bu-5.6.1
 for kernel 5.3.0-kali3-686-pae (i686)
2020年 04月 11日 星期六 00:40:15 CST
Makefile:2207: /usr/src/rtl88x2bu-5.6.1

/rtl8822b.mk: 没有那个文件或目录   # means 'no such file or diectory'
make: *** 没有规则可制作目标“/usr/src/rtl88x2bu-5.6.1  # means 'no rules could be used to make...'

/rtl8822b.mk”。 停止。  # means 'stop'

**How could I deal with it?
Thx a lot**

roy668899, 2020-04-10:
I'm pretty sure that rtl8822b.mk is in the directory

bevarb, 2020-05-11:
have you solved this problem?

metalheadcode, 2020-05-21:
i'm having same issue

metalheadcode, 2020-05-21:
[SOLVED] i just realize that this is old version, I try the new ver 5.6.1 it works! you can get [here](https://github.com/cilynx/rtl88x2bu)