cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #69: RPI4 compiling errors

davacker, 2020-05-17:
Hello, i bought a wifi antenna and I'm totally new to installing drivers on linux.
After doing a _lsusb_ I found my device ID, which is 

> 0bda:b812 Realtek Semiconductor Corp.

After some research I think I need rtl8812bu drivers.

So I went on my raspberry pi 4 and git clone this repo, change the platform in the Makefile and I couldn't install linux-headers from my kernel (4.19.97-v7l+), apt let me download 4.18.0-3-common
but not higher, I don't really know why.

Even without the good linux-headers I tried to make `dkms build -m rtl88x2bu -v ${VER}` and when I check in the log file here is what I get :

```
DKMS make.log for rtl88x2bu-5.3.1 for kernel 4.19.97-v7l+ (armv7l)
Sun 17 May 20:00:54 CEST 2020
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.19.97-v7l+/build M=/var/lib/dkms/rtl88x2bu/5.3.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.97-v7l+'
Makefile:614: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.19.97-v7l+'
make: *** [Makefile:2011: modules] Error 2
```

If someone can help me, it'd be very nice, 

Sorry for my bad English, 

Hope we can find a fix.

rrafal1337, 2020-05-17:
Hi. I think that your kernel headers package is ok. Try to solve this error with command:
```
sudo ln -s /lib/modules/$(uname -r)/build/arch/arm /lib/modules/$(uname -r)/build/arch/armv7l
```

davacker, 2020-05-17:
Thank you, you solved it !!

closed 2020-05-17 by davacker