cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #73: struct sha256_state compile error with Ubuntu 20.10

rolandbreedveld, 2020-11-06:
Cant compile this with Ubuntu 20.20:

#  dkms build -m rtl88x2bu -v ${VER}

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.8.0-26-generic KVER=5.8.0-26-generic src=/usr/src/rtl88x2bu-5.3.1...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl88x2bu: 5.3.1 not found
Error! Bad return status for module build on kernel: 5.8.0-26-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.3.1/build/make.log for more information.
root@roland:/usr/src/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959# cat /var/lib/dkms/rtl88x2bu/5.3.1/build/make.log
DKMS make.log for rtl88x2bu-5.3.1 for kernel 5.8.0-26-generic (x86_64)
vr  6 nov 2020 16:54:51 CET
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.8.0-26-generic/build M=/var/lib/dkms/rtl88x2bu/5.3.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-26-generic'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o
In file included from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:78,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_cmd.c:17:
/usr/src/rtl88x2bu-5.3.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_cmd.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:78,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_debug.c:17:
/usr/src/rtl88x2bu-5.3.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_debug.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:78,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_security.c:17:
/usr/src/rtl88x2bu-5.3.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_security.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:78,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_io.c:47:
/usr/src/rtl88x2bu-5.3.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_io.c:47:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
/usr/src/rtl88x2bu-5.3.1/core/rtw_security.c:2191:13: error: conflicting types for ‘sha256_init’
 2191 | static void sha256_init(struct sha256_state *md)
      |             ^~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_security.c:17:
./include/crypto/sha.h:136:20: note: previous definition of ‘sha256_init’ was here
  136 | static inline void sha256_init(struct sha256_state *sctx)
      |                    ^~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_security.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o] Error 1
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o] Error 1
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o] Error 1
make[1]: *** [Makefile:1780: /var/lib/dkms/rtl88x2bu/5.3.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-26-generic'
make: *** [Makefile:2011: modules] Error 2

vpolinov, 2021-02-06:
" rolandbreedveld commented on Nov 6, 2020"

I see, cilynx is not in a hurry to propose a fix, not at all.

slicingmelon, 2021-04-20:
Hey I've got it working using rincat's driver fork
```
sudo git clone "https://github.com/RinCat/RTL88x2BU-Linux-Driver.git" /usr/src/rtl88x2bu-git
sudo sed -i 's/PACKAGE_VERSION="@PKGVER@"/PACKAGE_VERSION="git"/g' /usr/src/rtl88x2bu-git/dkms.conf
sudo dkms add -m rtl88x2bu -v git
sudo dkms autoinstall
```
```
$ uname -r
5.10.0

```
However, it does not see 5G networks, I am unable to connect to my AP.