cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959 #74: install failed

vpolinov, 2021-02-06:
me@ARMADILLO2:~/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959$ sudo dkms build -m rtl88x2bu -v ${VER}

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.8.0-41-generic KVER=5.8.0-41-generic src=/usr/src/rtl88x2bu-5.3.1....(bad exit status: 2)
ERROR (dkms apport): binary package for rtl88x2bu: 5.3.1 not found
Error! Bad return status for module build on kernel: 5.8.0-41-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.3.1/build/make.log for more information.

hoefler50, 2021-02-06:
Hello I have on raspian on a raspberry 3 follow errors:

sudo dkms build -m rtl88x2bu -v 5.3.1

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.10.11-v7+ KVER=5.10.11-v7+ src=/usr/src/rtl88x2bu-5.3.1....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.11-v7+ (armv7l)
Consult /var/lib/dkms/rtl88x2bu/5.3.1/build/make.log for more information.

I think ist is the sam promlem. Have you a fix?
Thanks.
best regards Michael

gruentee, 2021-03-06:
Same here, running 

```Linux moode 5.10.20-v7+ #1404 SMP Thu Mar 4 19:40:23 GMT 2021 armv7l GNU/Linux```

on a RaspberryPi 2.